TheSpaghettiDetective/moonraker-obico #1: Install instructions don't work

crashmaxx, 2021-11-03:
Under the "How to run" section of the Readme.md, the following line did not work on my RPi4. It might work if you are already developing the repo, but it doesn't work on a clean install.

git pull https://github.com/TheSpaghettiDetective/tsd-moonraker

I found that I instead had to use:

 git clone https://github.com/TheSpaghettiDetective/tsd-moonraker.git

raymondh2, 2021-11-03:
This will be fixed in future. We are still working on developing this.

crashmaxx, 2021-11-03:
Thanks for the reply. I know it's early alpha, just wanted to help by testing it out. Working pretty good so far! I can put in a pull request to update the Readme.md if you'd like.

For anyone wanting to test the alpha on MainsailOS 0.5.0 using a RPi4 or similar setup, here are install instructions that worked for me.

```
# requires python3; install python3 packages
sudo apt-get install python3 python3-pip python3-venv

# clone repo
git clone https://github.com/TheSpaghettiDetective/tsd-moonraker.git

# setup virtual environment
python3 -m venv tsd-moonraker/
cd tsd-moonraker
source venv/bin/activate
pip3 install -r requirements.txt

# fill in essential configuration
cp config.sample.ini config.ini
nano config.ini

# link printer (grab tsd auth token)
python3 -m tsd_moonraker.link -c config.ini

# start app
python3 -m tsd_moonraker.app -c config.ini -l tsd_moonraker.log
```

raymondh2, 2021-12-15:
Fixed in #2

closed 2021-12-15 by raymondh2