TheSpaghettiDetective/moonraker-obico #3: Enable compatibility mode

Amacfa, 2021-12-20:
Thanks for the work on this plugin. It works great on a MacBook M1 Pro under arm architecture. 

The only thing that I couldn’t get working is a frame rate that matched my fluidd stream. It appears TSD server would only update the stream when I refreshed the page.  (Maybe it’s working on snapshots/basic mode)

Is there a command or change to make that can enable compatibility mode like with the full octoprint plugin?

Scoobler, 2021-12-21:
I have tried using both stream and snapshot - but I only get an image on refreshing the TSD page - the same happens with the temps, if you watch during the start of a print, you only get to see the temps change when you refresh the page.

After two prints, I have one which has a time-lapse - it appears the only images in the time-lapse are possibly from when I refreshed the page - although it was only a short print ~20mins.

Amacfa, 2021-12-21:
> I have tried using both stream and snapshot - but I only get an image on refreshing the TSD page - the same happens with the temps, if you watch during the start of a print, you only get to see the [...]

A temporary fix is to install a browser addon to auto refresh the page, thus triggering capturing ea h time it refreshed.

raymondh2, 2021-12-21:
We are aware of this issue.

Scoobler, 2021-12-21:
Thanks for the advice on the browser temp fix!

I guess the TSD would need this so it actually gets images to check for spaghetti?

Is this the same reason the iOS TSD app shows blank values for time remaining, % completed etc:
[image]
That’s 3ish hours into a 6 hour print.
Along with controls from the app not working (movements don’t happen when trying the controls in the app)? 
Although I did test cancel and that worked fine!

raymondh2, 2021-12-21:
Controls aren't implemented yet

raymondh2, 2021-12-22:
Closed in prior commit.

closed 2021-12-22 by raymondh2