TheSpaghettiDetective/moonraker-obico #9: Handshake status 404 Not Found

CraXyOW3, 2021-12-28:
Hi, I am getting an error about Handshake error.

```
(tsd-moonraker) pi@ender3:~/tsd-moonraker $ python3 -m tsd_moonraker.app -c config.ini -l tsd_moonraker.log
2021-12-28 14:33:30,024      INFO  main - starting tsd_moonraker (v0.1)
2021-12-28 14:33:30,028   WARNING  main.moonrakerconn - api key is unset, trying to fetch one
2021-12-28 14:33:30,066      INFO  main.moonrakerconn.ws - connecting to ws://127.0.0.1:7125/websocket
2021-12-28 14:33:30,087      INFO  main.moonrakerconn - waiting for klipper ready
2021-12-28 14:33:30,112      INFO  main.moonrakerconn - printer objects: ['webhooks', 'configfile', 'mcu', 'firmware_retraction', 'gcode_move', 'print_stats', 'virtual_sdcard', 'menu', 'display_status', 'output_pin beeper', 'pause_resume', 'filament_motion_sensor motion_sensor', 'probe', 'bed_mesh', 'heaters', 'heater_bed', 'tmc2209 stepper_x', 'tmc2209 stepper_y', 'tmc2209 stepper_z', 'tmc2209 extruder', 'fan', 'heater_fan extruder_cooler_fan', 'gcode_macro START_PRINT', 'gcode_macro END_PRINT', 'gcode_macro PRIME_LINE', 'gcode_macro CG28', 'gcode_macro G29', 'gcode_macro M125', 'gcode_macro LOW_TEMP_CHECK', 'gcode_macro pause_print', 'gcode_macro resume_print', 'gcode_macro CANCEL_PRINT', 'motion_report', 'query_endstops', 'idle_timeout', 'system_stats', 'toolhead', 'extruder']
2021-12-28 14:33:30,198      INFO  main - linked printer: {'is_pro': True, 'id': 2, 'name': 'Ender3'}
2021-12-28 14:33:30,199      INFO  main.tsdconn.ws - connecting to wss://MYDOMAINNAME/ws/dev/
2021-12-28 14:33:30,275     ERROR  main.tsdconn - Handshake status 404 Not Found (Handshake status 404 Not Found), reconnecting
2021-12-28 14:33:30,276     ERROR  main.tsdconn - backing off 0.196355 seconds: Handshake status 404 Not Found
2021-12-28 14:33:30,364      INFO  main.moonrakerconn - heaters: ['heater_bed', 'extruder']
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/tsd-moonraker/tsd_moonraker/wsconn.py", line 241, in start
    self.flow()
  File "/home/pi/tsd-moonraker/tsd_moonraker/app.py", line 131, in flow
    self.wait_for(self._received_last_job)
  File "/home/pi/tsd-moonraker/tsd_moonraker/wsconn.py", line 295, in wait_for
    if self._wait_for(event, process_fn, timeout_msecs):
  File "/home/pi/tsd-moonraker/tsd_moonraker/app.py", line 153, in _wait_for
    event, process_fn, timeout_msecs)
  File "/home/pi/tsd-moonraker/tsd_moonraker/wsconn.py", line 334, in _wait_for
    if process_fn(event):
  File "/home/pi/tsd-moonraker/tsd_moonraker/app.py", line 213, in _received_last_job
    Event(sender=self.id, name='last_job', data=jobs[0])
IndexError: list index out of range

2021-12-28 14:33:30,626      INFO  main.tsdconn.ws - connecting to wss://MYDOMAINNAME/ws/dev/
2021-12-28 14:33:30,627      INFO  main - linked printer: {'is_pro': True, 'id': 2, 'name': 'Ender3'}
2021-12-28 14:33:30,686     ERROR  main.tsdconn - Handshake status 404 Not Found (Handshake status 404 Not Found), reconnecting
2021-12-28 14:33:30,687     ERROR  main.tsdconn - backing off 0.691136 seconds: Handshake status 404 Not Found
2021-12-28 14:33:31,537      INFO  main - linked printer: {'is_pro': True, 'id': 2, 'name': 'Ender3'}
2021-12-28 14:33:31,537      INFO  main.tsdconn.ws - connecting to wss://MYDOMAINNAME/ws/dev/
2021-12-28 14:33:31,591     ERROR  main.tsdconn - Handshake status 404 Not Found (Handshake status 404 Not Found), reconnecting
2021-12-28 14:33:31,592     ERROR  main.tsdconn - backing off 0.633144 seconds: Handshake status 404 Not Found
2021-12-28 14:33:32,383      INFO  main - linked printer: {'is_pro': True, 'id': 2, 'name': 'Ender3'}
2021-12-28 14:33:32,384      INFO  main.tsdconn.ws - connecting to wss://MYDOMAINNAME/ws/dev/
2021-12-28 14:33:32,513     ERROR  main.tsdconn - Handshake status 404 Not Found (Handshake status 404 Not Found), reconnecting
2021-12-28 14:33:32,514     ERROR  main.tsdconn - backing off 1.466053 seconds: Handshake status 404 Not Found
2021-12-28 14:33:34,165      INFO  main.tsdconn.ws - connecting to wss://MYDOMAINNAME/ws/dev/
2021-12-28 14:33:34,165      INFO  main - linked printer: {'is_pro': True, 'id': 2, 'name': 'Ender3'}
2021-12-28 14:33:34,234     ERROR  main.tsdconn - Handshake status 404 Not Found (Handshake status 404 Not Found), reconnecting
2021-12-28 14:33:34,235     ERROR  main.tsdconn - backing off 5.936400 seconds: Handshake status 404 Not Found
```

This is with latest fluidd, followed the install instructions, successfully linked printer.
running latest tsd software (selfhosted, could this be the problem?) on another machine.

encetamasb, 2022-01-29:
Sorry, I have just noticed this open issue. 

Last job error is probably a bug and/or missing [history] section in moonraker config.

The other one is interesting, will try to reproduce/fix both next week.

encetamasb, 2022-01-31:
The first error, related to history/jobs is fixed.

The handshake error might be a sign of that your private server configuration is broken. Are your websocket connections routed correctly? Please test your private server with the octoprint plugin first. Can you see the same problem with that?

closed 2022-05-31 by raymondh2