TheSpaghettiDetective/moonraker-obico #14: Create ~/klipper_logs if it doesn't exist

mylesagray, 2022-02-08:
After installation, the systemd unit will fail to start if the directory for logs doesn't exist, as soon as it's created it runs successfully.

Maybe a handler for creating the folder for logs if it doesn't exist would be a good addition?

crashmaxx, 2022-02-09:
I agree making the directory if it doesn't exist is a good addition. Shouldn't be hard to add.

I'm curious though. Did it prompt you to input the log directory when you installed it? Did you go with the default even though your klipper logs are in a different place?

mylesagray, 2022-02-09:
No from what I remember there was no prompt so I just went with the defaults.

I use MainsailOS as my distro, and it doesn’t have a ~/klipper_logs folder, so only after digging through journalctl did I find out that’s what it was trying to do.

Sent from mobile

crashmaxx, 2022-02-09:
#15 created to make sure the log directory exists when installing.

encetamasb, 2022-02-09:
> #15 created to make sure the log directory exists when installing.

Merged. Thank you!

closed 2022-02-09 by encetamasb