TheSpaghettiDetective/moonraker-obico #16: Print progress not showing

raymondh2, 2022-02-09:
Print progress is stuck at 0% in TSD

encetamasb, 2022-02-10:
Could you send me the logs?

encetamasb, 2022-02-10:
Anyway, in mainsail the logic is lot more complex, I will check. https://github.com/mainsail-crew/mainsail/blob/86d3421f1ebb63da2112e76f6e735a874fda58bc/src/store/farm/printer/getters.ts#L71

encetamasb, 2022-02-10:
Does this commit solve it for you? https://github.com/TheSpaghettiDetective/tsd-moonraker/commit/0ac1a76b3b383dd81685c8f2a40ad03dc9f6ddaf

encetamasb, 2022-02-11:
Progress calculation based on virtual_sdcard module https://github.com/TheSpaghettiDetective/tsd-moonraker/commit/fb1b4fdbf0c9581264aad0d742f16b1c896498d7 like in Mainsail

encetamasb, 2022-02-22:
@raymondh2 Could you confirm that it works now? If yes, let's tag a release.

encetamasb, 2022-02-22:
Merged.

closed 2022-02-22 by encetamasb