TheSpaghettiDetective/moonraker-obico #24: Print progress not showing

scuk23, 2022-04-23:
I have tried reinstalling several times with no luck. I cannot see the progress of the print. See attached picture. It looks like an issues has been opened on this in the past, but it is not resolved for me. Thanks!

[image]

kennethjiang, 2022-04-26:
Thank you for submitting the bug. I'm currently busy with a few other priorities. If you can help me debug a bit more that'll be great!

kennethjiang, 2022-04-26:
No problem. Will get to it when I have a chance.

obs945, 2022-06-20:
My situation is the same as scuk23. 
Print progress percentage not showing issue also caused live AI detection and time-lapse not working. 
No matter I directly upload gcode to obico web interface or to moonraker, the situation is the same. 
The most strange part is the webcam working good and the panel showed "Looking Good" (but the pointer always stayed there without jiggle).

vinridge, 2022-06-20:
I am having the same exact same issues as obs945 and scuk23. I was using Octoprint an everything worked fine on Obico. 

I also noticed that when the printer is not printing, I cannot control the extruder or bed movement like i could with Octoprint. 
I have included the Obico log file from Klipper. Hope it helps!

I know the Klipper integration is still in beta but i figured I'd post these issues as well. I absolutely love this app and it's saved me from wasting filament on many of occasions. Keep up the good!

[moonraker-obico-7125.log](https://github.com/TheSpaghettiDetective/moonraker-obico/files/8942815/moonraker-obico-7125.log)

closed 2022-06-23 by kennethjiang