TheSpaghettiDetective/moonraker-obico #34: Only run ffmpeg when printer is printing

voidtrance, 2022-06-29:
The ffmpeg capture is started on app startup, which results in the ffmpeg process taking about 22% of CPU time on a Raspberry Pi4. However, the capture is not required until the printer is actually printing.

The app already receives events from Moonraker/Klipper and one of those events is "Printing". It would be nice if the ffmpeg capture is started when the printer transitions from a non-printing state to "Printing" and stopped when printing is done.

raymondh2, 2022-06-29:
Ffmpeg is how the 25 fps stream is created. So it is required outside of printing.

voidtrance, 2022-06-29:
> Ffmpeg is how the 25 fps stream is created. So it is required outside of printing.

Is there some place where I can read up on this because this doesn't make sense to me. Why would the stream be needed when not printing? What is the server doing with those images?

voidtrance, 2022-07-05:
> @voidtrance We used to stream only when printing. It became one of the most requested feature to stream the webcam when not printing. You may not have a use case for seeing the webcam when your printer is idle. But plenty of people do.
> 
> You can [disable this behavior](https://www.obico.io/docs/user-guides/disable-25-fps-streaming/#should-i-disable-the-webcam-streaming) if you don't feel comfortable.
> 
> I'm closing this issue as this is an intended behavior.

I'd like to re-open this issue since the suggested fix did not work.

The Klipper version of the Obico client does not have an interface to disable streaming. However, I set `disable_video_streaming` in `moonraker-obico.cfg` to `True`. After restarting the Moonraker-Obico service, ffmpeg is still being started.

kennethjiang, 2022-07-05:
`disable_video_streaming` not being effective sounds like a bug. I'm reopening this issue to reflect the it.

voidtrance, 2022-07-05:
Not that familiar with the code but from a simple read, is the config setting even being set in the `WebcamConfig` object?

```
        webcam_config = WebcamConfig(
            snapshot_url=config.get(
                'webcam', 'snapshot_url',
                fallback=''),
            snapshot_ssl_validation=config.getboolean(
                'webcam', 'snapshot_ssl_validation',
                fallback=False
            ),
            stream_url=config.get(
                'webcam', 'stream_url',
                fallback='http://127.0.0.1:8080/?action=stream'
            ),
            flip_h=config.getboolean(
                'webcam', 'flip_h',
                fallback=False
            ),
            flip_v=config.getboolean(
                'webcam', 'flip_v',
                fallback=False
            ),
            rotate_90=config.getboolean(
                'webcam', 'rotate_90',
                fallback=False
            ),
            aspect_ratio_169=config.getboolean(
                'webcam', 'aspect_ratio_169',
                fallback=False
            )
        )
```
To me this looks like the default value `    disable_video_streaming: bool = False` is being used.

voidtrance, 2022-07-06:
> Not that familiar with the code but from a simple read, is the config setting even being set in the `WebcamConfig` object?
>
> ```
>         webcam_config = WebcamConfig( [...]

Adding the following to `config.py` seems to stop ffmpeg from starting up when the service is started. I still have to confirm whether images are streamed properly when printing starts:
```
--- config.py.orig	2022-07-06 11:53:06.431749551 -0700
+++ config.py	2022-07-06 11:52:40.118657764 -0700
@@ -207,6 +207,10 @@
                 'webcam', 'aspect_ratio_169',
                 fallback=False
             ),
+            disable_video_streaming=config.getboolean(
+                'webcam', 'disable_video_streaming',
+                fallback=False
+            ),
         )
 
         logging_config = LoggingConfig(
```

commit 57022503 (2022-07-07): Closes #34
closed 2022-07-07 by kennethjiang