TheSpaghettiDetective/moonraker-obico #37: Explicitly set webcam stream/snapshot urls are ignored if retrieved from moonraker

BradenM, 2022-07-25:
Hi,

Currently, moonraker-obico appears to ignore explicitly set webcam snapshot/stream URLs if it succeeds in retrieving the values from moonraker. 

Given the following config (in my environment):
```
[moonraker]
host = moonraker
port = 7125

[webcam]
disable_video_streaming = False
snapshot_url = http://webcam:8080/snapshot
stream_url = http://webcam:8080/stream
```

This plugin succeeds in retrieving a _different_ URL from moonraker and proceeds to override my values.

Moonraker responds with a relative paths such as: `/webcam/?action=stream`, and moonraker-obico ignores my values and tries to connect to `localhost/webcam/?action=stream`

For context, I am working on running this plugin in a docker container in my local setup based on https://github.com/mkuf/prind

### Workaround

Commenting out / removing the following lines:
https://github.com/TheSpaghettiDetective/moonraker-obico/blob/03ab792d7d4d24980786044171b3e8c30bf1ecbc/moonraker_obico/config.py#L72-L77
resolves the issue as it removes any consideration of values retrieved from moonraker. 

Of course, this works as a temporary fix for my personal docker image :smile:, but I think a proper solution would be always to use any webcam URLs that are explicitly set in the config (and if not set, continue with the current logic). 

Thanks!

kennethjiang, 2022-07-25:
Thank you for reporting this issue! It does look like a bug in the code. Will think about how to fix it.

closed 2022-08-23 by kennethjiang

aguinaldoabbj, 2023-02-21:
@BradenM Sorry to necrobump this issue. I'm also running Klipper within prind stack and I can't get Obico to work. Could you please share any guidance on this?

MallocArray, 2023-02-21:
@aguinaldoabbj Better to open a new issue, or check out the Discord channel to discuss your issue.
https://www.obico.io/docs/user-guides/contact-us-for-support/

commit 7debec9a (2022-08-21): Closes #37. The webcam settings in config file take precedence over the ones configured in Moonraker