TheSpaghettiDetective/moonraker-obico #39: Time Format Bug - Klipper w/ Moonraker

MaliciousCodez05, 2022-08-11:
[image]

Running Obico version 1.66.120 : b187 on Android 12 and the latest versions of Klipper / moonraker-obico on mainsail. The top printer in the screenshot is running on Octoprint, but the bottom printer is running on the following versions of Klipper/Mainsail + Moonraker.

klipper
v0.10.0-546-ga709ba43

moonraker
v0.7.1-625-g413e3b2

moonraker-obico
v0.5.0-0-g574a741

mainsail
v2.2.1

Please let me know if I can help provide any additional information!

kennethjiang, 2022-08-11:
Thanks! @MaliciousCodez05

kennethjiang, 2022-08-23:
@MaliciousCodez05 This is fixed in 0.5.1.

closed 2022-08-23 by kennethjiang