TheSpaghettiDetective/moonraker-obico #43: Possibility to power on mcu over obico

JonathanHeinz, 2022-08-28:
Currently it is not possible to control / restart a printer / mcu that is turned off. It should possible to turn on the printer over the moonraker power control.

kennethjiang, 2022-08-28:
This sounds like a good feature to have. Is there existing function in Moonraker that allows that? If so, please insert the links here so that I can take a look at them.

JonathanHeinz, 2022-08-28:
Moonraker Power API: [Link](https://moonraker.readthedocs.io/en/latest/web_api/#power-apis)

kennethjiang, 2022-08-28:
I didn't know that! That's awesome! This API  makes it so much easier to implement this feature.

Erech1676, 2022-09-07:
I turn on and off my printer through Octoprint with a kasa or tp Link plug in that controls a kp303 powerstrip.

Hr-Ravn, 2022-09-27:
That would be a welcoming feature for me! 
I have a 1 channel relay controlled by the Gpio pins of my pi, to turn the printer on/off. 
this is setup in my moonraker config, so if this could retrieve the state of that, it would be much better along with the Tunneling!

kennethjiang, 2026-01-31:
> +1 from me, this is still in high regards! :)

Maybe you can send this PR that is tested, and I will approve it as soon as it doesn't break existing functions? I've never bothered to set up a power control in Moonraker, so I don't have a way to test it.