TheSpaghettiDetective/moonraker-obico #45: Obico eating a lot of memory until pi die

appleimperio, 2022-09-12:
ERROR  obico.utils - giving up on error: Janus quit! This should not happen. Exit code: 127
[image]

I verified the H-top and is in fact moonraker-obico. Since I have it installed through kiauh I think that might be the problem I uninstalled and installed directly with the same result. The only reason that I can think of might be an issue is that im running klipper ,moonraker and obico in 1 pi and mainsail in another one.

kennethjiang, 2022-09-12:
This is concerning.

Can you post an htop screenshot? Also does this happen when you are not printing?

appleimperio, 2022-09-12:
Sorry I just uninstalled and it only happened when printing. I have a top screenshot but it doesn't show the obico service just read python3

appleimperio, 2022-09-25:
I do two main things to resolve the issue. I don't know which one do the trick. First change the Pi from wifi to Ethernet and the other one since i have another pi running klipper but instead of using mjpeg streamer im using ustreamer. I change it to this one too, and it's working perfectly.

closed 2022-09-25 by appleimperio