TheSpaghettiDetective/moonraker-obico #48: "Not running on a pi" - any options for video?

robotfishe, 2022-10-08:
I'm running klipper on an Odroid C4 - partly due to the shortage of Raspberry Pis the last couple of years, but also because I've found the USB more stable.

Obico doesn't seem to like it. I can't get a video stream to load on my self-hosted Obico server, and when I look at the logs, I see "Not running on a Pi. Quiting video_pipeline." It seems that because I'm running a different SBC it's not even trying to load the video components. Is there any way around this behaviour? My webcam works fine and I can view the stream through Fluidd, I just can't get Obico to look for it.

kennethjiang, 2022-10-11:
It's an expected and documented behavior that the video streaming is only supported on Raspberry Pis: https://obico.io/docs/user-guides/webcam-stream-stuck-at-1-10-fps/#2-is-octoprint-running-on-a-raspberry-pi

The failure detection will work correctly on any platform.

I'm closing this issue as it's an expected behavior.

closed 2022-10-11 by kennethjiang

raptordemon, 2023-01-16:
this doesn't work on the BTT CB1 which is not a pi but is running debian, according to the docs it might work, is there a work around for this?
OS images that may support the webcam streaming
All MainsailOS versions older than 0.6.0.
All FluiddPi versions older than 1.17.0.
32-bit Raspberry Pi OS versions older than Buster.
All Raspbian versions.
**All other Debian derivatives, such as DietPi.**

raptordemon, 2023-01-16:
my issue was solved by fixing the django setup. its working now with the CB1. i was self hosted.

raptordemon, 2023-02-16:
i don't recall exactly what i did wrong but there was an issue with the standard django setup, i ran thru the django setup steps and it fixed it