TheSpaghettiDetective/moonraker-obico #50: error on install.sh and verification code not accepted.

andreklug, 2022-12-11:
HI, I have set up a self hosted Obico server running behind a Nginx proxy as described in the docs. I can access this server via https://obico.home. I also configured the domain as "https://obico.home" in django, tried without https:// before.
When I try to add a printer, I followed the instructions to clone the repo, then ran ./install.sh

Before that, I copied the moonraker-obico.cfg.sample to moonraker-obico.cfg and updated as folllows:

`[server]
url = https://obico.home
# auth_token: <let the link command set this, see more in readme>
# sentry_opt: out or in

[moonraker]
host = 127.0.0.1
port = 7125
# api_key = <grab one or set trusted hosts in moonraker>

[webcam]
disable_video_streaming = False

# flip_h = False
# flip_v = False
# rotate_90 = False
# aspect_ratio_169 = False

# CAUTION: Don't modify the settings below unless you know what you are doing
#   In most cases webcam configuration will be automatically retrived from moonraker
#
# snapshot_url = http://127.0.0.1:8080/?action=snapshot
# stream_url = http://127.0.0.1:8080/?action=stream

[logging]
path = /home/pi/klipper_logs/moonraker-obico.log
level = INFO

`

When running ./install.sh, this is the console:

`pi@ratos:~/moonraker-obico $ ./install.sh

======================================================================================================
###                                                                                                ###
###                       Install and Configure Obico for Klipper                                  ###
###                                                                                                ###
======================================================================================================

###### Installing required system packages... You may be prompted to enter password.

Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 https://deb.nodesource.com/node_16.x buster InRelease         
Hit:3 http://raspbian.raspberrypi.org/raspbian buster InRelease     
Hit:4 https://dl.yarnpkg.com/debian stable InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3 is already the newest version (3.7.3-1).
python3-venv is already the newest version (3.7.3-1).
ffmpeg is already the newest version (7:4.1.10-0+deb10u1+rpt1).
python3-pip is already the newest version (18.1-5+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

###### Detecting the softwares and settings of your Klipper system ...

**Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.7/configparser.py", line 958, in __getitem__
    raise KeyError(key)
KeyError: 'server'**

We couldn't automatically detect the settings. Please enter them below to continue.

!!!WARNING: Manually entering the Moonraker settings can be error prone.
We highly recommend using KIAUH if you have a non-standard Klipper installation, e.g., running multiple Moonraker instances.

Moonraker port: 7125^C
   ____
  / __ \
 | |  | | ___   ___   ___   ___  _ __  ___
 | |  | |/ _ \ / _ \ / _ \ / _ \| '_ \/ __|
 | |__| | (_) | (_) | (_) | (_) | |_) \__ \  _   _   _
  \____/ \___/ \___/ \___/ \___/| .__/|___/ (_) (_) (_)
                                | |
                                |_|

The installation has run into an error:

Installation interrupted by user or for unknown error.

Please fix the error above and re-run this setup script:

-------------------------------------------------------------------------------------------------
cd ~/moonraker-obico
./install.sh
-------------------------------------------------------------------------------------------------

Need help? Stop by:

- The Obico's help docs: https://obico.io/help/
- The Obico community: https://obico.io/discord/
`
The error does not appear when I run "sudo ./install.sh", but also then the verification code is not accepted. 

When opening the URL https://obico.home/api/v1/octo/verify/ in the browser I get a message "{"code":["This field may not be null."]}" - but this may be expected.

WHat can I do to successfully link my printer?

kennethjiang, 2022-12-12:
Can you run `bash -x ./install.sh  >~/obico-install.out` and then attach the ~/obico-install.out file here?

I suspect this is an environment issue. You can get help a lot more quickly if you come to our discord server: https://obico.io/discord/

andreklug, 2022-12-13:
Hi @kennethjiang 
Here is the resulting file:
[obico-install.txt](https://github.com/TheSpaghettiDetective/moonraker-obico/files/10215748/obico-install.txt)

Thank you!

kennethjiang, 2022-12-13:
It looks like the error was not captured in the file. Can you run this `bash -x ./install.sh >~/obico-install.out 2>&1` instead and attached the file again?

kennethjiang, 2022-12-14:
Can you confirm:

1. The Moonraker config file is at this path: `/home/pi/printer_data/config/moonraker.conf`
2. There is a `[server]` section in `/home/pi/printer_data/config/moonraker.conf`

andreklug, 2022-12-14:
1. yes
2. there was no [server] section in moonraker.conf. I have added it like this: with host and port, and the error above is gone. it detects the settings now. 

The code still is not accepted. 

Could this be related to SSL? I am using my own CA, do I need to add a trusted certificate or can I disable verification if this matters?

closed 2022-12-14 by kennethjiang