TheSpaghettiDetective/moonraker-obico #51: Error updating moonraker-Obico

ciapamusche, 2022-12-12:
As title said , Obico not update in klipper

The box in moonraker remain ILLEGAL. No way to recovery 

The output of recovery command says 

fatal: unable to access 'https://hub.fastg
it.xyz/TheSpaghettiDetective/moonraker-obi
co.git/': Failed to connect to hub.fastgit
.xyz port 443: Connection timed out

Same thing if I try to update by kiauh

How can I do?

Thank you

kennethjiang, 2022-12-13:
It looks like your cloned the repo from https://hub.fastgit.xyz/TheSpaghettiDetective/moonraker-obico.git/

Please clone it from https://github.com/TheSpaghettiDetective/moonraker-obico.git instead.

See detailed instructions here: https://obico.io/docs/user-guides/klipper-setup/

ciapamusche, 2022-12-13:
i installed by kiauh and the update manger file have the correct repo . Following procedure the result doesn't change... I can install but I cannot update.

`[update_manager moonraker-obico]
type: git_repo
path: ~/moonraker-obico
origin: https://github.com/TheSpaghettiDetective/moonraker-obico.git
env: /home/pi/moonraker-env/bin/python
requirements: requirements.txt
install_script: install.sh
managed_services:
  moonraker-obico`

So the issue still persist

kennethjiang, 2022-12-14:
Can you run this command and post the console output here?

`cd ~/moonraker-obico && git remote -v`

ciapamusche, 2022-12-14:
```
pi@speeder-pad:~/moonraker-obico$ cd ~/moonraker-obico && git remote -v                                                                                             origin  https://hub.fastgit.xyz/TheSpaghettiDetective/moonraker-obico.git (fetch                                                                                    )
origin  https://hub.fastgit.xyz/TheSpaghettiDetective/moonraker-obico.git (push)

```

after executing the official guide

ciapamusche, 2022-12-14:
`
pi@speeder-pad:~$ sudo git clone https://github.com/TheSpaghettiDetective/moonraker-obico.git
[sudo] password for pi:
Cloning into 'moonraker-obico'...
remote: Enumerating objects: 1388, done.
remote: Counting objects: 100% (657/657), done.
remote: Compressing objects: 100% (181/181), done.
remote: Total 1388 (delta 506), reused 603 (delta 467), pack-reused 731
Receiving objects: 100% (1388/1388), 25.69 MiB | 360.00 KiB/s, done.
Resolving deltas: 100% (904/904), done.
pi@speeder-pad:~$ git clone https://github.com/TheSpaghettiDetective/moonraker-obico.git
fatal: destination path 'moonraker-obico' already exists and is not an empty directory.
pi@speeder-pad:~$ cd moonraker-obico
pi@speeder-pad:~/moonraker-obico$ sudo ./install.sh

======================================================================================================
###                                                                                                ###
###                       Install and Configure Obico for Klipper                                  ###
###                                                                                                ###
======================================================================================================

###### Installing required system packages... You may be prompted to enter password.

Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Get:2 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
Get:5 https://pkgs.tailscale.com/stable/ubuntu focal InRelease
Get:6 http://ports.ubuntu.com/ubuntu-ports focal-updates/main armhf Packages [1189 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe armhf Packages [828 kB]
Fetched 2137 kB in 8s (253 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3 is already the newest version (3.8.2-0ubuntu2).
python3-venv is already the newest version (3.8.2-0ubuntu2).
ffmpeg is already the newest version (7:4.2.7-0ubuntu0.1).
python3-pip is already the newest version (20.0.2-5ubuntu1.6).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

###### Creating python virtual environment for moonraker-obico...

created virtual environment CPython3.8.10.final.0-32 in 1763ms
  creator CPython3Posix(dest=/root/moonraker-obico-env, clear=False, global=True)
  seeder FromAppData(download=False, pip=latest, setuptools=latest, wheel=latest, pkg_resources=latest, via=copy, app_data_dir=/root/.local/share/virtualenv/seed-app-data/v1.0.1.debian.1)
  activators BashActivator,CShellActivator,FishActivator,PowerShellActivator,PythonActivator,XonshActivator

###### Detecting the softwares and settings of your Klipper system ...

We couldn't automatically detect the settings. Please enter them below to continue.

!!!WARNING: Manually entering the Moonraker settings can be error prone.
We highly recommend using KIAUH if you have a non-standard Klipper installation, e.g., running multiple Moonraker instances.

Moonraker port: 7125
Moonraker config file: /home/pi/klipper_config/printer_1/moonraker.conf
Klipper log directory: /home/pi/klipper_logs/

================================= Obico Server URL ==============================================

Now tell us what Obico Server you want to link your printer to.
You can use a self-hosted Obico Server or the Obico Cloud. For more information, please visit: https://obico.io.
For self-hosted server, specify "http://server_ip:port". For instance, http://192.168.0.5:3334.

The Obico Server (Don't change unless you are linking to a self-hosted Obico Server): https://app.obico.io

###### Creating config file /home/pi/klipper_config/printer_1/moonraker-obico.cfg ...

###### Creating python virtual environment for moonraker-obico...
[... 42 lines cut ...]
MMMMMMMMMMMMMMMMMMMMMMMMMMMNd.      '0MMMMMMMMMMMMMMMMMMMMNl      :0WMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMMMMW0c.     oWMMMMMMMMMMMMMMMMMMMWl    ,xNMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMW0l.   .kWMMMMMMMMMMMMMMMMMMWl .;xXMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMWXkc'.,OWMMMMMMMMMMMMMMMMMWOdKWMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMN0xONMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM
MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM

====================================================================================================
###                                                                                              ###
###                                       SUCCESS!!!                                             ###
###                             Now enjoy Obico for Klipper!                                     ###
###                                                                                              ###
====================================================================================================

The changes we have made to your system:

- System service: /etc/systemd/system/moonraker-obico
- Config file: /home/pi/klipper_config/printer_1/moonraker-obico.cfg
- Update file:
- Inserted "[include moonraker-obico-update.cfg]" in the "moonraker.conf" file
- Log file:
`

This is the history of my actions

ciapamusche, 2022-12-14:
UPDATE

The issue seems solved.

I found  (I don't know why) a file .git in my pi directory called .gitconfig

inside was present this line:

[url "https://hub.fastgit.xyz/"]
	insteadOf = https://github.com/

deleted the file and reinstalled obico  the update manager seems to work.

kennethjiang, 2022-12-17:
Great. Thank you for sharing what you found out. I'll close the issue now.

closed 2022-12-17 by kennethjiang

ciapamusche, 2022-12-17:
Thank you to you.

Please note the .gticonfig file is hidden if browse with an explorer e.g cyberduck. So. Check the option to show hidden files