TheSpaghettiDetective/moonraker-obico #52: No image in obico

iqubik, 2022-12-14:
Cant see anything in obico site after last update
[moonraker-obico.log](https://github.com/TheSpaghettiDetective/moonraker-obico/files/10231453/moonraker-obico.log)

kennethjiang, 2022-12-17:
Please set the logging level to DEBUG and download the log file again. https://obico.io/docs/user-guides/moonraker-obico/logging-file/

Also this doesn't look like a bug. So the best way to get help quickly is to come to our discord server: https://obico.io/discord/

iqubik, 2023-01-05:
i have some problem
see video in mainsall, relink, delete, add again printer but still no video

Log:
2023-01-05 20:45:01,751      INFO  obico.app - Fetching linked printer...
2023-01-05 20:45:02,162      INFO  obico.app - starting moonraker-obico (v1.1.3)
2023-01-05 20:45:02,163      INFO  obico.app - Linked printer: {'is_pro': False, 'id': 130573, 'name': 'My Awesome Cloud Printer'}
2023-01-05 20:45:02,190   WARNING  sarge - No process found for Command('v4l2-ctl --list-devices')
2023-01-05 20:45:02,266      INFO  obico.app - Starting webcam streamer
2023-01-05 20:45:02,277   WARNING  obico.moonraker_conn - api key is unset, trying to fetch one
2023-01-05 20:45:02,280   WARNING  obico.webcam_stream - Not running on a Pi. Quiting video_pipeline.
2023-01-05 20:45:02,296   WARNING  obico.janus - No external Janus gateway. Not on a Pi. Skipping Janus connection.
2023-01-05 20:45:02,322      INFO  obico.moonraker_conn - connection is open
2023-01-05 20:45:02,548      INFO  obico.moonraker_conn - Klippy ready
2023-01-05 20:45:02,798      INFO  obico.app - detected state change: Offline -> Operational
2023-01-05 20:45:11,220      INFO  backoff - Backing off capture_jpeg(...) for 0.8s (requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.0.203', port=80): Max retries exceeded with url: /webcam/?action=photo (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffffa82e2700>: Failed to establish a new connection: [Errno 113] No route to host')))
2023-01-05 20:45:14,285      INFO  backoff - Backing off capture_jpeg(...) for 1.0s (requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.0.203', port=80): Max retries exceeded with url: /webcam/?action=photo (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffffa82d9100>: Failed to establish a new connection: [Errno 113] No route to host')))
2023-01-05 20:45:17,356     ERROR  backoff - Giving up capture_jpeg(...) after 3 tries (requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.0.203', port=80): Max retries exceeded with url: /webcam/?action=photo (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffffaab7ec40>: Failed to establish a new connection: [Errno 113] No route to host')))
2023-01-05 20:45:17,358     ERROR  obico.utils - 
Traceback (most recent call last):
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/connection.py", line 174, in _new_conn
    conn = connection.create_connection(
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/util/connection.py", line 95, in create_connection
    raise err
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/util/connection.py", line 85, in create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/connectionpool.py", line 398, in _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/connection.py", line 239, in request
    super(HTTPConnection, self).request(method, url, body=body, headers=headers)
  File "/usr/lib/python3.9/http/client.py", line 1255, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.9/http/client.py", line 1301, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.9/http/client.py", line 1250, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.9/http/client.py", line 1010, in _send_output
    self.send(msg)
  File "/usr/lib/python3.9/http/client.py", line 950, in send
    self.connect()
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/connection.py", line 205, in connect
    conn = self._new_conn()
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/connection.py", line 186, in _new_conn
    raise NewConnectionError(
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object at 0xffffaab7ec40>: Failed to establish a new connection: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/requests/adapters.py", line 439, in send
    resp = conn.urlopen(
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/connectionpool.py", line 787, in urlopen
    retries = retries.increment(
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/urllib3/util/retry.py", line 592, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.0.203', port=80): Max retries exceeded with url: /webcam/?action=photo (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffffaab7ec40>: Failed to establish a new connection: [Errno 113] No route to host'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
[... cut ...]
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/home/orangepi/moonraker-env/lib/python3.9/site-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.0.203', port=80): Max retries exceeded with url: /webcam/?action=photo (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xffffaab7ec40>: Failed to establish a new connection: [Errno 113] No route to host'))

iqubik, 2023-01-05:
> 

can fix this by add strings:
[webcam]
disable_video_streaming = False
snapshot_url = http://127.0.0.1:8080/?action=snapshot
stream_url = http://127.0.0.1:8080/?action=stream

kennethjiang, 2023-01-10:
Glad you fixed this problem yourself. Closing it now.

closed 2023-01-10 by kennethjiang