TheSpaghettiDetective/moonraker-obico #55: Moonraker Warning(s) on start up, everything else working properly.

VashBaldeus, 2023-01-29:
The warnings:

> **Moonraker warning**
> 
> Unparsed config option 'managed_services:
> moonraker-obico' detected in section [update_manager moonraker-obico]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
> 
> **Moonraker warning**
> 
> Unparsed config option 'install_script: install.sh' detected in section [update_manager moonraker-obico]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
> 
> **Moonraker warning**
> 
> Unparsed config option 'requirements: requirements.txt' detected in section [update_manager moonraker-obico]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
> 
> **Moonraker warning**
> 
> Unparsed config option 'origin: https://github.com/TheSpaghettiDetective/moonraker-obico.git' detected in section [update_manager moonraker-obico]. This may be an option no longer available or could be the result of a module that failed to load. In the future this will result in a startup error.
> 
> **Moonraker warning**
> 
> [update_manager]: Failed to load extension moonraker-obico: [Errno 13] Permission denied: '/root/moonraker-obico-env/bin/pip'

Regardless of these issues, Obico connects to my self-hosted Obico server and works normally.

zellneralex, 2023-02-01:
Can you share your moonraker.log, that looks more like an config error

VashBaldeus, 2023-02-02:
> Can you share your moonraker.log, that looks more like an config error

https://pastebin.com/ziNhRZyP

zellneralex, 2023-02-02:
ok that is not normal 
```
[update_manager moonraker-obico]
type: git_repo
path: ~/moonraker-obico
origin: https://github.com/TheSpaghettiDetective/moonraker-obico.git
env: /root/moonraker-obico-env/bin/python
requirements: requirements.txt
install_script: install.sh
managed_services:	moonraker-obico
```
Did you try to install obico as root?

zellneralex, 2023-02-02:
I would expect it would look like 
```
[update_manager moonraker-obico]
type: git_repo
path: ~/moonraker-obico
origin: https://github.com/TheSpaghettiDetective/moonraker-obico.git
env: ~/moonraker-obico-env/bin/python
requirements: requirements.txt
install_script: install.sh
managed_services: moonraker-obico
```
But for that I need to see your file system

kennethjiang, 2023-02-02:
@zellneralex Good catch! I think it's because the user installed moonraker-obico as root. @VashBaldeus Please confirm if you can see this message.

VashBaldeus, 2023-02-03:
> @zellneralex Good catch! I think it's because the user installed moonraker-obico as root. @VashBaldeus Please confirm if you can see this message.

kennethjiang, 2023-02-07:
@VashBaldeus Can you confirm if you installed moonraker-obico as root?

VashBaldeus, 2023-02-08:
> @VashBaldeus Can you confirm if you installed moonraker-obico as root?
I installed it as `pi` user on my raspberry pi. when running the install command I applied `sudo`, could this be the cause of the issue? if so, how do i re-install?

kennethjiang, 2023-02-15:
> > @VashBaldeus Can you confirm if you installed moonraker-obico as root?
> > I installed it as `pi` user on my raspberry pi. when running the install command I applied `sudo`, could this be the cause of the issue? if so, how do i re-install?

This could be the issue. Yes please re-install without `sudo`. The script will run sudo to elevate the access when/if necessary.

VashBaldeus, 2023-02-18:
> This could be the issue. Yes please re-install without `sudo`. The script will run sudo to elevate the access when/if necessary.

Okay, wish I knew how to uninstall it, since i am not super linux savvy.