TheSpaghettiDetective/moonraker-obico #59: no video with 1.2.1

synfinatic, 2023-02-28:
Video and heater bed/raspi temp sensors don't seem to work with v1.2.1.  Reverting back to v1.2.0 seems to resolve the issue.

Logs showing me restarting v1.2.1 and it failing followed by trying v1.2.0 which works:

```
2023-02-27 19:31:48,307      INFO  obico.app - starting moonraker-obico (v1.2.1)
2023-02-27 19:31:48,307      INFO  obico.app - Linked printer: {'is_pro': True, 'id': 1, 'name': 'Voron v2.040'}
2023-02-27 19:31:48,541      INFO  obico.app - Starting webcam streamer
2023-02-27 19:31:48,548   WARNING  obico.moonraker_conn - api key is unset, trying to fetch one
2023-02-27 19:31:48,613      INFO  obico.moonraker_conn - connection is open
2023-02-27 19:31:48,723      INFO  obico.moonraker_conn - Klippy ready
2023-02-27 19:31:48,972      INFO  obico.app - detected state change: Offline -> Operational
2023-02-27 19:31:49,576      INFO  backoff - Backing off wait_for_janus(...) for 1.0s (ConnectionRefusedError: [Errno 111] Connection refused)
2023-02-27 21:40:37,993      INFO  backoff - Backing off capture_jpeg(...) for 0.3s (AttributeError: 'ServerConn' object has no attribute 'snapshot_url')
2023-02-27 21:40:38,270      INFO  backoff - Backing off capture_jpeg(...) for 1.1s (AttributeError: 'ServerConn' object has no attribute 'snapshot_url')
2023-02-27 21:40:39,413     ERROR  backoff - Giving up capture_jpeg(...) after 3 tries (AttributeError: 'ServerConn' object has no attribute 'snapshot_url')
2023-02-27 21:54:48,339      INFO  obico.app - detected state change: Operational -> Printing
2023-02-27 21:54:48,462      INFO  obico.app - print event: PrintStarted (1677534888)
2023-02-27 23:17:59,816      INFO  obico.app - detected state change: Printing -> Operational
2023-02-27 23:17:59,817      INFO  obico.app - print event: PrintDone (1677534888)
2023-02-27 23:34:49,774      INFO  obico.app - detected state change: Operational -> Printing
2023-02-27 23:34:49,880      INFO  obico.app - print event: PrintStarted (1677540889)
2023-02-27 23:38:19,345      INFO  obico.app - detected state change: Printing -> Paused
2023-02-27 23:38:19,345      INFO  obico.app - print event: PrintPaused (1677540889)
2023-02-27 23:39:39,109      INFO  obico.app - detected state change: Paused -> Operational
2023-02-27 23:39:39,110      INFO  obico.app - print event: PrintCancelled (1677540889)
2023-02-27 23:39:39,614      INFO  obico.app - print event: PrintFailed (1677540889)
2023-02-27 23:46:48,037      INFO  obico.app - detected state change: Operational -> Printing
2023-02-27 23:46:48,139      INFO  obico.app - print event: PrintStarted (1677541607)
2023-02-28 01:06:15,356      INFO  obico.app - Fetching linked printer...
2023-02-28 01:06:15,447      INFO  obico.app - starting moonraker-obico (v1.2.1)
2023-02-28 01:06:15,448      INFO  obico.app - Linked printer: {'is_pro': True, 'id': 1, 'name': 'Voron v2.040'}
2023-02-28 01:06:15,655      INFO  obico.app - Starting webcam streamer
2023-02-28 01:06:15,670   WARNING  obico.moonraker_conn - api key is unset, trying to fetch one
2023-02-28 01:06:15,746      INFO  obico.moonraker_conn - connection is open
2023-02-28 01:06:15,882      INFO  obico.moonraker_conn - Klippy ready
2023-02-28 01:06:16,136      INFO  obico.app - detected state change: Offline -> Printing
2023-02-28 01:06:16,724      INFO  backoff - Backing off wait_for_janus(...) for 1.0s (ConnectionRefusedError: [Errno 111] Connection refused)
2023-02-28 01:08:17,557      INFO  obico.app - Fetching linked printer...
2023-02-28 01:08:17,612      INFO  obico.app - starting moonraker-obico (v1.2.0)
2023-02-28 01:08:17,613      INFO  obico.app - Linked printer: {'is_pro': True, 'id': 1, 'name': 'Voron v2.040'}
2023-02-28 01:08:17,742      INFO  obico.app - Starting webcam streamer
2023-02-28 01:08:17,755   WARNING  obico.moonraker_conn - api key is unset, trying to fetch one
2023-02-28 01:08:17,824      INFO  obico.moonraker_conn - connection is open
2023-02-28 01:08:18,023      INFO  obico.moonraker_conn - Klippy ready
2023-02-28 01:08:18,276      INFO  obico.app - detected state change: Offline -> Printing
2023-02-28 01:09:50,347      INFO  obico.app - detected state change: Printing -> Operational
2023-02-28 01:09:50,353      INFO  obico.app - print event: PrintDone (1677541607)
```

kennethjiang, 2023-03-03:
Can you set logging level to DEBUG and post the log file here? https://obico.io/docs/user-guides/moonraker-obico/logging-file/

synfinatic, 2023-03-07:
here ya go [moonraker-obico.log.zip](https://github.com/TheSpaghettiDetective/moonraker-obico/files/10914156/moonraker-obico.log.zip)

kennethjiang, 2023-03-09:
There is no obvious errors in the log file. Can you join our discord server https://obico.io/discord and post the question there as I will need to do some real time troubleshooting with you?

kennethjiang, 2023-07-09:
Closing as I believe this issue is solved

closed 2023-07-09 by kennethjiang