TheSpaghettiDetective/moonraker-obico #61: Moonraker-Obico pause stop the bed heater

Caouette1988, 2023-03-22:
My Obico is set to stop only the extruder heater when pausing a print when it detect a fail, but it also stop the bed heater.
The turn off bed heater was unchecked.
Any thoughts?
Thanks

MallocArray, 2023-03-22:
How long was it sitting idle?  I recently discovered that the macro for M600 from the macro sample keeps the default idle timeout of 10 minutes and if you don't resume in that time, it turns off all heaters and disables the steppers so you can't resume.  This could be the same issue
I added this to my M600 macro to extend the timeout, you may want to add it to the Pause macro instead, if that is what Obico calls when a failure is detected
```
SET_IDLE_TIMEOUT TIMEOUT=6000
```

Caouette1988, 2023-03-22:
You're right, I have a couple macro to turn off heater and printer after idle time and also after the extruder temp get below 35.
There is a way to keep these macro and keep obico pause feature?
Thank you.

kennethjiang, 2023-03-23:
This is a known issue - almost all modern printers have a heater timeout now. Not sure if there is a good way to solve this problem. Please let me know if you have suggestions.

Caouette1988, 2023-03-24:
> How long was it sitting idle? I recently discovered that the macro for M600 from the macro sample keeps the default idle timeout of 10 minutes and if you don't resume in that time, it turns off all [...]

I find out it not only turn off all heater, but disable stepper too after 10mins. So you were right, bacause here my idle timout setting :
[idle_timeout]
gcode:
   TURN_OFF_HEATERS
   M84             ; disable steppers
timeout: 600      ;10 min

I'm reading back your response, I had not understand the first time, now I think I get it.
I need to specify a M600 macro with the command SET_IDLE_TIMEOUT TIMEOUT=6000 in it. 6000 stand for 100 minutes. 
After that 100 minutes, steppers et heaters will be disable by the idle_timeout.

So I supose that obico send a M600 command to pause the printer when there is possible failed?

Is the synthaxes correct and do I need to define more line in the macro?

[gcode_macro M600]
gcode:
    SET_IDLE_TIMEOUT TIMEOUT=6000

MallocArray, 2023-03-24:
In my use case I was trying to do a filament change and M600 is the correct gcode for that, but ultimately it was calling the PAUSE macro, so I think you may want to add the idle_timeout increase to the Pause feature, as I assume that is what Obico is calling, but hopefully @kennethjiang can clarify if that is what it does when Obico detects a failure

I'm using Mainsail, so inside of mainsail.cfg I have the original macro:
```
[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
gcode:
  PAUSE_BASE
  _TOOLHEAD_PARK_PAUSE_CANCEL {rawparams}
```

I'm thinking you may want to define this again in your own macro file or printer.cfg, since mainsail.cfg is read-only so you can redefine it after your printer.cfg line `[include mainsail.cfg]` (or I assume fluidd has something similar)

Then you can add the value you see fit, such as
```
[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
gcode:
  SET_IDLE_TIMEOUT TIMEOUT=6000
  PAUSE_BASE
  _TOOLHEAD_PARK_PAUSE_CANCEL {rawparams}
```

I haven't done this myself, but it is what I plan on doing.

Caouette1988, 2023-03-24:
Perfect, more logical that it use Pause.
I'll try that and keep you updated.

I was reading the mainsail.cfg
I saw that firmware retraction can be enabled while pausing.
There is a way to add this to obico pausing?

Caouette1988, 2023-03-24:
Here the line command about fw retraction that are already in mainsail.cfg : 

  {% set use_fw_retract = False if not macro_found
                     else False if client.use_fw_retract is not defined
                     else True  if client.use_fw_retract|lower == 'true' and printer.firmware_retraction is defined
                     else False %}

Caouette1988, 2023-03-25:
What I finaly do : 

`[delayed_gcode delayed_pause_hotend_off]
initial_duration: 300. #(5 mins)
gcode:
  {% if printer.pause_resume.is_paused %}
    SET_HEATER_TEMPERATURE HEATER=extruder TARGET=0
  {% endif %}

[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
gcode:
  SET_GCODE_VARIABLE MACRO=RESUME VARIABLE=last_extruder_temp VALUE="{printer[printer.toolhead.extruder].target}"
  PAUSE_BASE
  _TOOLHEAD_PARK_PAUSE_CANCEL {rawparams}
  SET_IDLE_TIMEOUT TIMEOUT=7200 ; 2hrs
  UPDATE_DELAYED_GCODE ID=delayed_pause_hotend_off DURATION=300 #(5 mins)

[gcode_macro RESUME]
description: Resume the actual running print
rename_existing: RESUME_BASE
variable_last_extruder_temp: 0
gcode:
  SET_IDLE_TIMEOUT TIMEOUT=600  
  ##### get user parameters or use default #####
  {% set macro_found = True if printer['gcode_macro _CLIENT_VARIABLE'] is defined else False %}
  {% set client = printer['gcode_macro _CLIENT_VARIABLE'] %}
  {% set velocity = printer.configfile.settings.pause_resume.recover_velocity %}
  {% set sp_move        = velocity if not macro_found else client.speed_move|default(velocity) %}
  ##### end of definitions #####
  SET_HEATER_TEMPERATURE HEATER=extruder TARGET={last_extruder_temp}
  TEMPERATURE_WAIT SENSOR=extruder MINIMUM={last_extruder_temp*0.98} MAXIMUM={last_extruder_temp*1.02}
  _CLIENT_EXTRUDE
  RESUME_BASE VELOCITY={params.VELOCITY|default(sp_move)}`

But this guy's method interest me, I think this would be more reliable
`[idle_timeout]
gcode:
{% if printer.pause_resume.is_paused %}
  {action_respond_info("Extruder powered down on idle timeout.")}
  M117 Idle Timeout, hotend cooldown
  M104 S0 # Set Hot-end to 0C (off)
{% else %}
  M117 Idle Timeout, poweroff
  TURN_OFF_HEATERS
  M84
{% endif %}`
I find this there: [https://github.com/mainsail-crew/mainsail-config/pull/13](url)

But with thus method, temp still need to be restaured before resume print

kennethjiang, 2023-03-26:
Obico is using the standard Moonraker API to pause the print when a failure is detected (or you press the "pause" button in the Obico app). The API we use is: https://moonraker.readthedocs.io/en/latest/web_api/#pause-a-print