TheSpaghettiDetective/moonraker-obico #62: It takes too long until Print is paused

BoBdBmstr, 2023-03-22:
When obico detects a failure, it takes between 3 and 5 minutes before printing is actually paused. Is this a bug in the software? If possible, it would be nice to be able to decide for yourself after how long a failure is detected before printing is stopped or paused.

I run obico as a docker self hosted with moonraker.

kennethjiang, 2023-03-23:
Pausing a print in Obico should be exactly the same as pausing it in Mainsail/Fluidd, as we are using the same API to pause the print. Do you have the same problem when you try to pause a print in Mainsail/Fluidd?

BoBdBmstr, 2023-03-25:
When i try it in Mainsail it Just works fine. Eventually the Sensitivity für detection ist Just Not high enough for me. Because i would Stop the Print when obico says fishy for more then a Minute. It is already set to "high".

kennethjiang, 2023-03-26:
Oh I think I know what you mean. Unfortunately the AI can't detect a failure as soon as it appears. It will wait for a few minutes to make sure it's actually a failure. Otherwise, the AI may produce too many false alarms.

You can think of it as a bug. But it's not a problem that can be easily solved with the current state of AI. I'll close it for now.

closed 2023-03-26 by kennethjiang