TheSpaghettiDetective/moonraker-obico #64: Pre-compiled Janus not available for amd64

mhaas, 2023-04-10:
Hi,

I am running klipper on an amd64 thin client. It seems that the supplied janus executable is only compiled for arm64. Hence, I get the "can only stream at 0.1fps" warning.

kennethjiang, 2023-04-18:
This problem will be solved eventually by trying to install OS-provided janus.

Reubertt, 2023-05-12:
I'm having this same problem on a rpi-os running in VM. I didn't quite understand how to solve it, changing the OS?

MallocArray, 2023-05-12:
@Reubertt Obico checks to see if it is running on actual Raspberry Pi hardware to enable full video stream. If you are running in a VM, it will not pass this check and you'll get snapshots only, but that will not impact failure detection, only your ability to watch a live stream

Reubertt, 2023-05-12:
> @Reubertt Obico checks to see if it is running on actual Raspberry Pi hardware to enable full video stream. If you are running in a VM, it will not pass this check and you'll get snapshots only, but that will not impact failure detection, only your ability to watch a live stream

The problem is that due to this Janus error, Obico is not even recognizing the snapshot because it detects 0.1 FPS

kennethjiang, 2023-05-13:
Like what @MallocArray , Janus is not involved in failure detection. On the 0.1FPS is used for failure detection.

burak1000, 2023-09-18:
how can i install janus

kennethjiang, 2023-09-19:
If you are debian, you can install it with `sudo apt install janus`