TheSpaghettiDetective/moonraker-obico #66: BUG on install | unable to install

kg6ebj, 2023-04-14:
The following error happens every time I try to install Obico. 
Linux klipper 5.15.63-sunxi64 #trunk SMP PREEMPT Sat Sep 3 17:13:20 PDT 2022 aarch64 GNU/Linux

======================================================================================================
###                                                                                                ###
###                       Install and Configure Obico for Klipper                                  ###
###                                                                                                ###
======================================================================================================

###### Installing required system packages... You may be prompted to enter the password.

Hit:2 http://deb.volian.org/volian scar InRelease                              
Hit:3 http://inovato.net/apt stable InRelease                                  
Hit:4 http://security.debian.org bullseye-security InRelease                   
Hit:5 http://deb.debian.org/debian bullseye InRelease                          
Get:6 https://pkgs.tailscale.com/stable/debian bullseye InRelease              
Hit:7 http://deb.debian.org/debian bullseye-updates InRelease                  
Hit:8 http://deb.debian.org/debian bullseye-backports InRelease      
Hit:1 https://box64.debian.ryanfortner.dev/debian ./ InRelease                 
Fetched 6,016 B in 16s (378 B/s)               
Reading package lists... Done
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ffmpeg is already the newest version (7:4.3.5-0+deb11u1).
python3-pip is already the newest version (20.3.4-4+deb11u1).
python3-virtualenv is already the newest version (20.4.0+ds-2+deb11u1).
python3 is already the newest version (3.9.2-3).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

###### Creating python virtual environment for moonraker-obico...

ModuleNotFoundError: No module named 'virtualenv.activation.xonsh'

kennethjiang, 2023-04-18:
What SBC and the OS image did you use for your Klipper? @kg6ebj

kg6ebj, 2023-04-19:
> What SBC and the OS image did you use for your Klipper? @kg6ebj

It was on an Inovato Quadra using the default armbian install. I tried both with and without Kiauh.

The workaround I found was to manually generate the Python venv and then run the script.

Also, you should know that I changed to using my rPi 4B because of the lack of streaming fps on the server.  It just shows buffering; pulling the stream from <klipper IP>/webcam or an IP camera would be much easier instead of restreaming the feed from the crowsnest.

I tried a rPi 4B, intel 15-6xxx, and the quadra, and only the rPi had any stream capability, even though the intel has 60FPS on crowsnest stream.

kennethjiang, 2023-04-21:
For this particular issue, it looks like a defective system package: https://stackoverflow.com/questions/71086270/no-module-named-virtualenv-activation-xonsh

For the RPi streaming issue, you can get help on Obico's discord server: https://obico.io/discord

closed 2023-04-21 by kennethjiang