TheSpaghettiDetective/moonraker-obico #83: Moonraker-obico not captured stream

YaphetS1, 2023-11-14:
Hi there!

Can some one help me? I'd install Obico API and client success. All in Docker, except the camera-streamer it run as systemctl daemon. Config of moonraker-obico has that configuration.

```
[server]
url = http://192.168.1.30:3334 # host machine with static IP in docker container
auth_token = xxxxxxxxxxxxxxx

[moonraker] # Connected
host = 192.168.1.30 # host machine with static IP in docker container
port = 7125 # public port

[webcam] # Conf IPs of local machine where it's daemon run
disable_video_streaming = False
snapshot_url = http://192.168.1.30:8080/snapshot
stream_url = http://192.168.1.30:8080/stream

[logging]
path = /opt/printer_data/logs/moonraker-obico.log
level = INFO

[tunnel]
```

Configuration of containers is 

```
CONTAINER ID   IMAGE                             COMMAND                  CREATED          STATUS          PORTS                                       NAMES
606fd6a14981   moonraker_obico                   "/opt/venv/bin/pytho…"   10 minutes ago   Up 10 minutes                                               moonraker-obico
464bc218e607   dimalo/fluidd:latest              "/docker-entrypoint.…"   9 hours ago      Up 7 hours      0.0.0.0:8010->80/tcp, :::8010->80/tcp       fluidd
77f340b58709   dimalo/klipper-moonraker:latest   "supervisord -c /etc…"   9 hours ago      Up 7 hours      0.0.0.0:7125->7125/tcp, :::7125->7125/tcp   klipper
a72212d4bc95   obico-server-web                  "sh -c 'python manag…"   9 hours ago      Up 9 hours      0.0.0.0:3334->3334/tcp, :::3334->3334/tcp   obico-server-web-1
15a63d661077   obico-server-tasks                "sh -c 'celery -A co…"   9 hours ago      Up 9 hours      3334/tcp                                    obico-server-tasks-1
098db0715c8f   obico-server-ml_api               "bash -c 'gunicorn -…"   9 hours ago      Up 9 hours      3333/tcp                                    obico-server-ml_api-1
b840a85322d4   redis:5.0-alpine                  "docker-entrypoint.s…"   9 hours ago      Up 9 hours      6379/tcp                                    obico-server-redis-1
```

But when printer is linked I saw only that message without capture stream in HOST/printers/N/control/
Meanwhile in klipper, I HAVE a camera stream. Configuration in moonraker.conf has similar...

```
rpi@rpi:~/Docker/klipper-web-control-docker $ sudo docker logs 606fd6a14981ffcaa67b12504b47e54b40ba517fee07c4dc370c5d5909d3b59c
2023-11-14 19:09:46,294      INFO  obico.app - Fetching linked printer...
2023-11-14 19:09:46,341      INFO  obico.app - Linked printer: {'is_pro': True, 'id': 3, 'name': 'My Awesome Cloud Printer'}
2023-11-14 19:09:46,359     ERROR  obico.nozzlecam - Failed to build nozzle config
Traceback (most recent call last):
  File "/opt/moonraker-obico/moonraker_obico/nozzlecam.py", line 57, in create_nozzlecam_config
    ext_info = self.server_conn.send_http_request('GET', f'/ent/api/printers/{self.printer_id}/ext/', timeout=60, raise_exception=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/moonraker-obico/moonraker_obico/server_conn.py", line 152, in send_http_request
    resp.raise_for_status()
  File "/opt/venv/lib/python3.12/site-packages/requests/models.py", line 953, in raise_for_status
    raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 404 Client Error: Not Found for url: http://192.168.1.30:3334/ent/api/printers/3/ext/
2023-11-14 19:09:46,367   WARNING  obico.moonraker_conn - api key is unset, trying to fetch one
2023-11-14 19:09:46,368   WARNING  obico.app - Connections not ready. Trying again in 1s...
2023-11-14 19:09:46,396      INFO  obico.moonraker_conn - connection is open
2023-11-14 19:09:46,664      INFO  obico.moonraker_conn - Klippy ready
2023-11-14 19:09:47,406   WARNING  obico.app - error response from moonraker, Event(name='message', data={'jsonrpc': '2.0', 'error': {'code': 400, 'message': "{'error': 'WebRequestError', 'message': 'Invalid argument'}"}, 'id': 30959}, sender='moonrakerconn')
2023-11-14 19:09:47,408      INFO  obico.app - detected state change: Offline -> Operational
2023-11-14 19:09:47,411      INFO  obico.janus - Starting webcam streamer
2023-11-14 19:09:47,424      INFO  obico.webcam_stream - Trying to start ffmpeg using camera-streamer H.264 source
2023-11-14 19:09:47,916     ERROR  obico.webcam_stream - STDOUT:
None
STDERR:
b'[tcp @ 0x5578c194f0] Connection to tcp://127.0.0.1:8080 failed: Connection refused\nhttp://127.0.0.1:8080/video.mp4: Connection refused\n'

2023-11-14 19:09:47,917      INFO  obico.webcam_stream - No camera-stream H.264 source found. Continue to legacy streaming: ffmpeg failed! Exit code: 1
```

YaphetS1, 2023-11-14:
1. In camera-steamer.py are hardcoded localhost? Is it worth?
2. I have try to change it at public ip of my machine (keep in mind it's run in container) but I don't have any success...
3. judging by the current logic, it still had to switch to the legacy mod, and the host and port from the config are already used there, but it doesn't or? 😰

YaphetS1, 2023-11-15:
I think the problem in obico-server...
Because in logs of camera-streamer all was good, it receive the connection between docker container with moonraker-obico and send it to that. And moonraker-obiko doesn't show any new errors in container. Meanwhile in camera-streamer logs all is good...
```
Nov 15 13:37:39 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/7: Client connected 172.17.0.2 (fd=4).
Nov 15 13:37:39 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/7: Request 'GET' '/snapshot' ''
Nov 15 13:37:39 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/7: Client disconnected 172.17.0.2.
Nov 15 13:37:39 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/1: Client connected 172.17.0.2 (fd=5).
Nov 15 13:37:39 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/1: Request 'GET' '/snapshot' ''
Nov 15 13:37:40 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/1: Client disconnected 172.17.0.2.
Nov 15 13:37:40 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/6: Client connected 172.17.0.2 (fd=13).
Nov 15 13:37:40 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/6: Request 'GET' '/snapshot' ''
Nov 15 13:37:40 rpi camera-streamer[524128]: util/http/http.c: HTTP8080/6: Client disconnected 172.17.0.2.
```

kennethjiang, 2023-11-17:
We use github issues only for bugs and feature requests. This issue doesn't look like a bug. Probably just configuration/routing issue.

You will get help more quickly by asking the question in our discord channel:
https://obico.io/discord-obico-klipper/

I'll close this issue for now. Again please discuss the issue in our discord channel. I will reopen this issue if it's confirmed to be a bug.

closed 2023-11-17 by kennethjiang

YaphetS1, 2023-11-20:
It's actual issue stream. The details in discord chanel [#obico](https://discord.com/channels/758059413700345988/1153007942208069692) [linkToMessage](https://discord.com/channels/758059413700345988/1153007942208069692/1176202802343723038)

In short "Bug in dokcer file or instruction or image of moonraker-obico"

YaphetS1, 2023-11-23:
@kennethjiang