TheSpaghettiDetective/moonraker-obico #85: Failed to build image pip during installation

Evo609, 2023-11-18:
I have a problem during installation of K1 script.

`###### Creating python virtual environment for moonraker-obico...

RuntimeError: failed to build image pip because:
Traceback (most recent call last):
  File "/opt/lib/python3.11/site-packages/virtualenv/seed/embed/via_app_data/via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
  File "/opt/lib/python3.11/site-packages/virtualenv/seed/embed/via_app_data/pip_install/base.py", line 33, in install
    self._uninstall_previous_version()
  File "/opt/lib/python3.11/site-packages/virtualenv/seed/embed/via_app_data/pip_install/base.py", line 144, in _uninstall_previous_version
    dist_name = self._dist_info.stem.split("-")[0]
                ^^^^^^^^^^^^^^^
  File "/opt/lib/python3.11/site-packages/virtualenv/seed/embed/via_app_data/pip_install/base.py", line 110, in _dist_info
    raise RuntimeError(msg)  # pragma: no cover
    ^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: no .dist-info at /root/.local/share/virtualenv/wheel/3.8/image/1/CopyPipInstall/pip-23.3.1-py3-none-any, has pip
`

kennethjiang, 2023-11-19:
There is only 1 K1 between all Obico team members. And we couldn't reproduce this problem on that K1. So it'll be great if @Evo609 , or anyone who can reproduce this problem on the K1, can do some debugging for us.

ArnoS146, 2024-05-14:
I got the same issue on the K1 Max

ArnoS146, 2024-05-16:
Has anyone had any luck on this? I tested it again on the K1 Max V2 still the same issue