TheSpaghettiDetective/moonraker-obico #90: K1 uninstall

Guilouz, 2024-02-04:
When installing Obico from installer, it install many package from entware:

[image]

Can you provide me a full list what it can be uninstalled safely when removing Obico ? To keep clean OS.

kennethjiang, 2024-02-04:
I believe all these were just python3 and python3-pip dependencies. The code is here:

https://github.com/TheSpaghettiDetective/moonraker-obico/blob/9a8640b506129df5f3e4fee8da1e39427607b9bb/scripts/install_creality.sh#L32

I think all of them can be removed if you are not nothing else is using python3 or python3-pip.

Guilouz, 2024-05-02:
@kennethjiang Do you have Discord? I need to talk to you about important thing.

kennethjiang, 2024-05-02:
@kennethjiang you can join our https://obico.io/discord/ and there you should be able to find me pretty easily.