TheSpaghettiDetective/moonraker-obico #93: buster-backports Release no longer has a Release file (Elegoo Neptune 4)

stevewade, 2024-04-15:
**Describe the bug**

**To Reproduce**
1. SSH into Elegoo Neptune 4 
2. Follow instructions on https://www.obico.io/blog/elegoo-neptune-4-and-obico-ai-3d-printing-revolution/
3. At `Step 4: Run the Installation Script` upon running the `./install.sh `command, the following error is seen:

`###### Installing required system packages... You may be prompted to enter password.

[sudo] password for mks: 
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Ign:3 http://deb.debian.org/debian buster-backports InRelease
Hit:4 http://security.debian.org buster/updates InRelease
Err:5 http://deb.debian.org/debian buster-backports Release
  404  Not Found [IP: 2a04:4e42:f::644 80]
Reading package lists... Done
E: The repository 'http://deb.debian.org/debian buster-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.`

[image]

kennethjiang, 2024-04-16:
Thank you @jhaury for providing the solution! This is the best option that I'm aware of. I believe this problem was caused by the fact that debian.org has dropped official support for Buster. 

Closing it as it's a 3rd-party problem, and it has a good workaround.

closed 2024-04-16 by kennethjiang

stevewade, 2024-04-17:
Worked like a charm, thank @kennethjiang