TheSpaghettiDetective/moonraker-obico #95: buster-backports not found

DrGrydou, 2024-04-23:
Hello,

I have an error while trying to install Obico on my Elegoo Neptune 4 plus.
I followed the steps mentioned [here](https://obico.io/blog/elegoo-neptune-4-and-obico-ai-3d-printing-revolution/#step-2-clone-the-obico-for-klipper-repository) but I get this after the command ./install.sh :
mks@mkspi:~/moonraker-obico$ ./install.sh

          ,----..
         /   /   \
        /   .     :    ,---,      ,--,
       .   /   ;.  \ ,---.'|    ,--.'|                  ,---.
      .   ;   /  ` ; |   | :    |  |,                  '   ,'\
      ;   |  ; \ ; | :   : :    `--'_        ,---.    /   /   |
      |   :  | ; | ' :     |,-. ,' ,'|      /     \  .   ; ,. :
      .   |  ' ' ' : |   : '  | '  | |     /    / '  '   | |: :
      '   ;  \; /  | |   |  / : |  | :    .    ' /   '   | .; :
       \   \  ',  /  '   : |: | '  : |__  '   ; :__  |   :    |
        ;   :    /   |   | '/ : |  | '.'| '   | '.'|  \   \  /
         \   \ .'    |   :    | ;  :    ; |   :    :   `----'
          `---`      /    \  /  |  ,   /   \   \  /
                     `-'----'    ---`-'     `----'

================> Obico for Klipper (Moonraker-Obico) <================
###                                                                 ###
###                 * AI-Powered Failure Detection                  ###
###              * Free Remote Monitoring and Access                ###
###               * 25FPS High-Def Webcam Streaming                 ###
###                   * Free 4.9-Star Mobile App                    ###
###                                                                 ###
=======================================================================

###### Installing required system packages... You may be prompted to enter password.

[sudo] password for mks:
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Ign:3 http://deb.debian.org/debian buster-backports InRelease
Hit:4 http://security.debian.org buster/updates InRelease
Err:5 http://deb.debian.org/debian buster-backports Release
  404  Not Found [IP: 2a04:4e42:6a::644 80]
Reading package lists... Done
E: The repository 'http://deb.debian.org/debian buster-backports Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

What is this Err:5 and how to continue the install ?

**Environment (please complete the following information):**
 - Printer make and model : Elegoo Neptune 4 Plus
 - Are you running Klipper on a Raspberry Pi? No
 - moonraker-obico version: The last one I guess...

billkenney, 2024-04-23:
I was actually going to open the same issue. You can get around the error by commenting that line out in the sources list, but this shouldn’t cause the install to fail. Here’s a workaround:
sudo sed -Ei 's/^(deb http:\/\/deb.debian.org\/debian buster-backports)/#\1/' /etc/apt/sources.list

kennethjiang, 2024-04-23:
Duplicate with https://github.com/TheSpaghettiDetective/moonraker-obico/issues/93

closed 2024-04-23 by kennethjiang