TheSpaghettiDetective/moonraker-obico #98: [BUG] k1 install failed MarkupSafe

Pokeylooted, 2024-05-12:
**Describe the bug**
The bug is upon installing this for Creality , when it tries to import flask the error` No module named `MarkupSafe` `

**To Reproduce**
Steps to reproduce the behavior:
1. Use the helper script, or run the install_creality.sh script
2. Wait till it gets to the linking section
3. View the error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - Printer make and model [e.g., Elegoo Neptune 4] Creality K1
 - Are you running Klipper on a Raspberry Pi? [Yes|No] No
 - moonraker-obico version: v0.8.0-354-gba942855

**Additional context**
This is mainly caused by python 3.8 installing wrong version of MarkupSafe, it will install markupsafe-0.0.0, instead of markupsafe-2.0.0 which python 3.11 installs perfectly on the same system. You can fix this by just copying the files manually into it which probably wouldn't be a recommended solution, however does work!

Pokeylooted, 2024-05-13:
On a clean install tried this again, Python 3.8 in sites-packages isn't generating a markupsafe folder. Only a MarkupSafe-0.0.0-py3.8.egg-info . Which seems to be the root cause of this issue.

kennethjiang, 2024-05-20:
Fixed in https://github.com/TheSpaghettiDetective/moonraker-obico/pull/99

closed 2024-05-20 by kennethjiang