TheSpaghettiDetective/moonraker-obico #105: [BUG] jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'

lettore, 2024-06-08:
Following [this guide ](https://www.obico.io/docs/user-guides/first_layer_ai/slicer-configuration-for-first-layer-ai-klipper-only/) , adding the suggested code in Orca Slicer and printing the print stop caused by error `!! Error evaluating 'gcode_macro _OBICO_LAYER_CHANGE:gcode': jinja2.exceptions.UndefinedError: 'dict object' has no attribute 'polygon'`, Orca Slicer and Moonraker Obico are at latest version.

kennethjiang, 2024-06-09:
Can you attach the gcode file to this issue?

I'm currently on vacation until 6/20. But I will take a look once I'm back.

lettore, 2024-06-09:
This is what I added in Orca Slicer
`_OBICO_LAYER_CHANGE CURRENT_LAYER={layer_num + 1} MINX=[first_layer_print_min_0] MINY=[first_layer_print_min_1] MAXX=[first_layer_print_max_0] MAXY=[first_layer_print_max_1]`
and this is the result in Gcode.
`_OBICO_LAYER_CHANGE CURRENT_LAYER=1 MINX=52.5576 MINY=27.0948 MAXX=173.881 MAXY=197.905`
no need to add the rest of Gcode as the error is just from this macro

kennethjiang, 2024-06-16:
I'm on vacation rn with very bad internet connection. Will try it when I'm back home next week.

kennethjiang, 2024-07-09:
Sorry for the delay. Finally got a chance to look at it. I tested a similar case with v2.0.2 but couldn't reproduce this problem.

If you are on an older version, can you update to v2.0.2 and test it again? @lettore