TheSpaghettiDetective/moonraker-obico #112: Provide pause parameters provided by obico app to PAUSE command

Matszwe02, 2024-12-08:
**Is your feature request related to a problem? Please describe.**
I want to handle obico pause a different way from a normal pause

**Describe the solution you'd like**
when obico's app is providing "retract when paused", "turn off heater when paused" etc. expose  them as `PAUSE`'s parameters. In that way someone can handle these parameters inside their printer to actually retract or turn off heaters, that work with their pause macro, but other may actually differentiate obico's pause from user's pause and perform obico specific pause gcode.

**Describe alternatives you've considered**
no alternatives exist

**Additional context**

kennethjiang, 2024-12-09:
> **Is your feature request related to a problem? Please describe.** I want to handle obico pause a different way from a normal pause
>
> **Describe the solution you'd like** when obico's app is [...]

When the system automatically pause the print due to a detected failure, it'll set

`initiator='system'`

https://github.com/TheSpaghettiDetective/obico-server/blob/99eda40cec9e82af745e945f4c67cfddc6e45d57/backend/api/octoprint_views.py#L235

Is this what you are looking for?

Matszwe02, 2024-12-09:
That might be it, but am I able to detect the initiator from klipper's PAUSE macro?

kennethjiang, 2024-12-10:
> That might be it, but am I able to detect the initiator from klipper's PAUSE macro?

It will require some code change in moonraker-obico to propagate this parameter to the macro.

If you can make the change and submit a PR, I'll be happy to review and merge it.

Matszwe02, 2024-12-10:
> > That might be it, but am I able to detect the initiator from klipper's PAUSE macro?
> 
> It will require some code change in moonraker-obico to propagate this parameter to the macro.
> 
> If you can make the change and submit a PR, I'll be happy to review and merge it.

Ok, thanks! I'll check out how it can be done. Do you suggest adding it as a parameter for the macro? What that parameter could be? Something like `PAUSE REASON=OBICO`?

kennethjiang, 2024-12-11:
> Ok, thanks! I'll check out how it can be done. Do you suggest adding it as a parameter for the macro? What that parameter could be? Something like `PAUSE REASON=OBICO`?

TBH I'm not super familiar with how Klipper macro works. maybe you can check with the Klipper community and propose a design here?

Matszwe02, 2024-12-11:
> TBH I'm not super familiar with how Klipper macro works. maybe you can check with the Klipper community and propose a design here?

It would be hard to implement a completly new functionality to klipper, which isn't really that useful. I suggest sending `PAUSE OBICO=1` instead of just `PAUSE` to moonraker, which will be easily read in the klipper's macro 
```py
{% if 'OBICO' in rawparams %}
...
```

and without that implementation, there will be no differences from the current pause command.

I'm familiar with klipper macros, so I can write some implementation and let you know how it's working. Alternatively I suggested on obico-app that the pause gcode could be customized, but they said it will be rarely used and not worth coding that.