TheSpaghettiDetective/moonraker-obico #116: [BUG] Brief bug description here

dralkaar, 2025-04-30:
On long back to back prints moonraker-obico doesn't clear ram and leads to high ram usage until 100%, that ultimately leads to printer shutdown.

**To Reproduce**
Steps to reproduce the behavior:
Print long prints back to back (without restarting the printer). I'm using the self hosted obico if that's important. 
It happened first 2 days ago (but i had no idea what happened at that pointy) and again today but luckily i was around the printer and caught the high ram usage warning on the printer screen. Running "top -c" in console revealed the issue (moonraker-obico was using over 80% at that point)

**Environment (please complete the following information):**
 - Printer make and model Voron 2.4R2
 - Klipper is running on BTT Pi V2
 - moonraker-obico version:  v2.0.11-4-gfdd48f3e

**Additional context**
I had no problem in the past so i suspect this is an issue that appeared after the last update.

OS: Armbian 25.2.3 bullseye
Distro: armbian
Load: 0.26, Mem: 484.3 MB / 1.9 GB, Temp: 36°C 

Current distro and load after stopping the moonraker-obico.

kennethjiang, 2025-05-03:
This is concerning. However, if overall ram usage is 484.3 MB / 1.9 GB, it doesn't look bad.

Can you post screenshots for

1. Upon reboot, what the "top -c" command shows?
2. When it runs for a while to a point when you believe moonraker-obico uses high memory,  what the "top -c" command shows?

dralkaar, 2025-05-05:
It seems that this issue was, probably, due to an os update. After that last OS update (armbian) now everything is working properly again and after running almost 2 full days back-to-back prints  the memory usage on moonraker-obico doesn't exceed 5%.

I will close this issue now since it seems it wasn't moonraker-obico related but OS related.

closed 2025-05-05 by dralkaar