TheSpaghettiDetective/moonraker-obico #117: [BUG] Obico causing timeout with Beacon probing

TheLion, 2025-05-21:
**Describe the bug**
I noticed timeouts with probing with my Beacon a lot. Recently reinstalled my RPi with a clean RatOS image. Didn't experience any timeouts on probing any more.

Yesterday I re-installed Obico on Moonraker and immediately the timouts happened again.

I stopped and disabled the Obico service on my RPi (with `sudo systemctl stop moonraker-obico.service` and 
`sudo systemctl disable moonraker-obico.service`) and the timeouts stopped.

Enabled and started the service again and the timeouts started happening again.

It doesn't happen first try every time with probing though and it seems to happen more often when the bed has to travel a largere distance.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Obico on Moonraker
2. Home Z a couple of times
3. ...

**Screenshots**

With Obico service enabled and started

[image]

With Obico service disabled and stopped

[image]

**Environment (please complete the following information):**
 - RatRig V-Coce-3 300 with Beacon Rev H (but it also happened with an earlier rev of the Beacon, replaced it because I thought it was faulty). Beacon is directly connected to the RPi by USB.
 - Are you running Klipper on a Raspberry Pi? [Yes]
 - moonraker-obico version: v2.0.11-4-gfdd48f3e

**Additional context**
Can't make sense why Obico would interfere with Beacon. Or any other Z-probes?

kennethjiang, 2025-05-22:
I can see 2 possible causes:

1. `[include moonraker_obico_macros.cfg]` in the printer.cfg file. Comment out this line in printer.cfg to see if it makes a difference.
2. moonraker-obico is using too much CPU and somehow this causes this timeout.  Use "top" to check how much moonraker-obico is using when enabled.

TheLion, 2025-05-22:
> I can see 2 possible causes:
>
>     1. `[include moonraker_obico_macros.cfg]` in the printer.cfg file. Comment out this line in printer.cfg to see if it makes a difference.
>
>     2. moonraker-obico is [...]

That sounds like a plausible explanation ;-)

1. Commenting out the include doesn't change anything.
2. Obico does not appear at the top with the service is started, but ffmpeg does appear at the top with a usage of 98/99% CPU usage when Obico service is started. When the service is stopped, ffmpeg does not exist in the top 10 of processes.

kennethjiang, 2025-05-22:
> That sounds like a plausible explanation ;-)
>
>     1. Commenting out the include doesn't change anything.
>
>     2. Obico does not appear at the top with the service is started, but ffmpeg does appear [...]

Ok CPU usage is the culprit. We have limited ffmpeg to use only 1 CPU, so for most users, it's not a problem. It sounds like the way beacon probe works makes it more susceptible to high CPU usage.

You can solve this problem by disabling streaming: https://www.obico.io/docs/user-guides/moonraker-obico/webcam/#disable_video_streaming

I will close this issue.

closed 2025-05-22 by kennethjiang

TheLion, 2025-05-29:
Even with streaming disabled, Z-probing just timed out, twice in a row. And CPU usage was not exceeding 40% at that moment.
Stopped Obico service and all went well.
Very strange.